SmartThingsCommunity/st-device-sdk-c #1: where to get the iot_device_prov_data information?

easion, 2019-11-08:
I compiled and run st-device-sdk-c/example on the openwrt operating system, but I need to find the information contained in iot_device_prov_data from the BSP NV store in order to make the program run. The documentation does not mention these.

YongsulOh, 2019-11-08:
Thanks easion & your feedback.

I will check it with our colleague and update to you again

DongHoon-Ryu, 2019-11-11:
Thank you for your trial.
In fact, such data comes from SmartThings Application during the onboarding process. That means we have to implement a SoftAP in Linux. But, unfortunately we are now planning to support SoftAP on a rich resource board such as Raspberry Pi because we were focusing on resource limited MCU devices first.
Therefore, please refer to the _es_wifiprovisioninginfo_handler() for resource limited MCU devices in the  "src/easysetup/iot_easysetup_d2d.c" file. I'll inform you again if there's anything updated about it.

Kwang-Hui, 2020-06-17:
As @DongHoon-Ryu said, provisioning data comes from SmartThings mobile application based on your Samsung account and test environment which includes WiFi access point SSID, end-point of SmartThings Cloud.
I hope this would be helpful to you.

Thanks.

closed 2020-06-17 by Kwang-Hui