SmartThingsCommunity/st-device-sdk-c #4: Support CMake

momo1, 2020-01-08:
Hi,

Is it possible to add CMake support? CMake works better with IDEs namely CLion. 

Thanks

DongHoon-Ryu, 2020-01-08:
Hi~,

Thank you for your suggestion.
We have also had a plan to support CMake. But we are not currently working on the relevant task because we are focusing more on increasing the applicable chipsets or operating systems. I think it would be very helpful if you could make a contribution related to CMake. Is that possible?

Thanks,
DongHoon Ryu

momo1, 2020-01-09:
Hi,

I would love to but I wouldn't be able to push it to your repo. Unfortunately I am not allowed to contribute to open source software outside of work without their approval and it's a hassle to get it approved.

Thanks

Kwang-Hui, 2020-01-09:
> Hi,
> 
> I would love to but I wouldn't be able to push it to your repo. Unfortunately I am not allowed to contribute to open source software outside of work without their approval and it's a hassle to get it approved.
> 
> Thanks

Hi momo1,
I'm using VScode for C project (actually just source editing).
I never use CLion before, Is it worth to pay subscription fee rather than use VScode?

> Hi,
> 
> Is it possible to add CMake support? CMake works better with IDEs namely CLion.
> 
> Thanks

momo1, 2020-01-09:
Hi,

100% worth it. I think you can try it for free. Just give yourself time to get over the switching learning curve then you'll really enjoy it.

Kwang-Hui, 2020-01-09:
> Hi,
> 
> 100% worth it. I think you can try it for free. Just give yourself time to get over the switching learning curve then you'll really enjoy it.

Great! Thanks for your comment.

Kwang-Hui, 2020-02-20:
I write CMakeLists.txt only for IDE testing
https://github.com/SmartThingsCommunity/st-device-sdk-c/pull/5
It may requires further validation

Kwang-Hui, 2020-04-20:
st-device-sdk-c supports cmake now.

closed 2020-04-20 by Kwang-Hui