SmartThingsCommunity/st-device-sdk-c #62: Potential segmentation fault

toddaustin07, 2020-11-10:
Hello!  I was exchanging messages with Kwang-Hui on the SmartThings Community boards.  I am working on a BSP for Raspberry Pi.  In my testing so far I uncovered a condition which can cause a segmentation fault in the iot_easysetup_http module.  

This is the condition:  if an unexpected GET request is received and it does not contain an expected command, then line 414 will cause a segmentation fault due to attempting strlen() on 'payload', which does not get a value set in this case (it is a pointer with value 0).

Regarding progress on getting RPI working with the example app:  I now have a SoftAP setup using hostapd and I can connect to it from my iphone.  However, I have not been successful getting the ST mobile app to connect.  I have confirmed that the SSID is configured to what you expect.  Are there any other requirements I need to know about?  I am running IOS 12.4.8 on iPhone6.  Can it scan and find the SoftAP automatically while connected to my wireless router?  Or do I have to manually change the wifi network?  (I thought this was improved in more recent IOS versions)  Does the ip address, wifi channel number, or auth mode matter?

I noticed that there are duplicate test devices showing in the ST mobile app even though there is only 1 in the developer workspace.  Is this an issue?  Also, I have seen mention in the documentation about the requirement of being a member of an organization for the APIs to work.  I am an individual developer, however I was assign an mmId in the developer workspace.  Could this be an issue?

If you would prefer that I use the community for these questions, I will post these questions there.

Thank you.

Todd Austin

Kwang-Hui, 2020-11-10:
Hi @toddaustin07 ,
Thanks for report issue.
The wrong usage of strlen() problem has fixed with https://github.com/SmartThingsCommunity/st-device-sdk-c/commit/ff6938c692bb2499b85eb159327faadb62f9d528#diff-8c0a6333d7c3674911908d993347d4c3e9355f20b806e633c7f22df0fdeda7d7 commit (it is still at `develop` branch. but will be merged master branch soon)

Regarding your problem on iPhone, iOS doesn't allow to get WiFi scanlist application privilege. ST app can't detect the existence of SoftAP device just like ST app at Android platform can do.
So it is highly recommended to create QR code (Please refer https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/master/doc/STDK_QR.md and great contribution from @grobebar  https://github.com/SmartThingsCommunity/st-device-sdk-c/tree/master/tools/qrgen) and use ST app v1.6.51.XX or later.

1. Turn on developer mode
2. Select my testing device or scan QR
3. ST app will ask user to get confirm connect to device's SoftAP.

Thank you.

Kwang-Hui Cho

toddaustin07, 2020-11-11:
Hi.  I tried using the QR code when provisioning as you recommended, but I'm still unable to connect for some reason.  I double-checked the SSID and it is correct.  Here is the error screen I'm getting on the mobile app:

[image]

I have version 1.6.52-437 of the ST mobile app.  I can connect to the access point and use it normally through iphone settings, so I know Hostapd is working OK.  But for some reason the ST app can't connect.  The SoftAP is set up with SSID=Todds_RPI_Tes_E4fE4b0016q_iSl2FQ; broadcasting the SSID; wifi channel 6, open authorization (no password).

Any other thoughts?

Kwang-Hui, 2020-11-11:
Hi @toddaustin07 ,

I think you can try with other SoftAP configuration refering https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/56a509640153d7937d395e678a32c30ab9d34a0b/src/iot_api.c#L142

toddaustin07, 2020-11-11:
Yes!  I changed authentication and used that password and now it connects. 

But why is this password hardcoded in the SDK??  How would device manufacturers know to use it?  I would suggest this requirement needs to be specified in the BSP porting documentation.

One more question:  can you explain the purpose of IOT_WIFI_MODE_SCAN and what is supposed to happen?  How does setting mode to Scan differ from calling iot_bsp_wifi_get_scan_result ?  Perhaps it's obvious to someone working with a more traditional SOC, but more explanation in the porting guide would be helpful.   :-)

Thank you!

toddaustin07, 2020-11-11:
OK, next question (thank you for being patient with me - I'm making progress...)

There are several provisioning data files that the SDK appears to require on local storage:  WifiProvStatus, IotAPSSID, ServerPort, etc.   I was assuming that I did not have to create or initialize these - that your code manages them.  However, I'm running into a file open error for ./IotAPPASS.  It appears the other files are getting created but not this one.  

My configuration is that my wireless router here does NOT require a password.  Is this causing a problem?  I'm not talking about the SoftAP (I've gotten past those connection issues); I'm talking about the access point you connect to in station mode.   I can select the AP from the list displayed on the mobile AP, so I know it gets that far.  But then it stopped because of the password file open error, and now I get it during initialization as well.  WifiProvStatus=DONE, but the password file is not there. 

Do you assume there must be wpa/wpa2 password on the AP?

Kwang-Hui, 2020-11-11:
> Yes! I changed authentication and used that password and now it connects.
> 
> But why is this password hardcoded in the SDK?? How would device manufacturers know to use it? I would suggest this requirement needs to be specified in the BSP porting documentation.
> 
> Thank you!

Hi @toddaustin07 ,

We expected that porting layer handle it with given parameters of `iot_bsp_wifi_set_mode()`
 >I would suggest this requirement needs to be specified in the BSP porting documentation.
Thanks for your suggestion.

Kwang-Hui, 2020-11-11:
> OK, next question (thank you for being patient with me - I'm making progress...)
> 
> There are several provisioning data files that the SDK appears to require on local storage: WifiProvStatus, IotAPSSID, ServerPort, etc. I was assuming that I did not have to create or initialize these - that your code manages them. However, I'm running into a file open error for ./IotAPPASS. It appears the other files are getting created but not this one.
> 
> My configuration is that my wireless router here does NOT require a password. Is this causing a problem? I'm not talking about the SoftAP (I've gotten past those connection issues); I'm talking about the access point you connect to in station mode. I can select the AP from the list displayed on the mobile app, so I know it gets that far. And it knows password is not required for that AP because the password field is disabled/greyed-out.
> 
> But then the SDK core code fails with the error. WifiProvStatus=DONE, but the password file is not there. I assume that it should exist, but be blank for my situation (open authorization). Perhaps this is a bug?

Let me have

toddaustin07, 2020-11-11:
Hi Kwang-Hui - 

Wanted to let you know that I successfully completed the provisioning process and the example device app (switch) is now working on my RPI!

To get around the bug I reported last night I temporarily changed my wireless router authentication to require a password.  But I would assume you should also support authentication mode=OPEN, so hope you can make a correction to the code and handle the case where there is no password.  I was looking at iot-nv_data.c and figured either to change the set wifi provisioning function to write a blank file if password length is 0, or have the get function ignore a missing password file and set it to null.  But I wasn't sure of implications to the rest of the code, so I didn't try to change anything myself.

Kwang-Hui, 2020-11-11:
> Hi Kwang-Hui -
>
> Wanted to let you know that I successfully completed the provisioning process and the example device app (switch) is now working on my RPI!
>
> To get around the bug I reported last [...]

Wow great. Congratulations!

I'll look into SDK code what you mentioned. Thanks again.

toddaustin07, 2020-11-21:
Hello again Kwang-Hui - I hope you are doing well.  I am continuing to refine the wifi provisioning automation for my Raspberry Pi-based direct connect device.  I have run into some error conditions that I'm trying to figure out, but I need some clarification on something:  Can you please explain in detail what is expected when Scan mode is requested?  What are the specific tasks you expect of the wifi module and what are the inputs and outputs?  I don't think I'm clear on this and how it is related to the get_scan_result function which is separately called.  I've looked at the esp32 code but it is equally unclear what "scan mode" does and how it is different from getting the scan results.

gamsahabnida

Kwang-Hui, 2020-11-23:
Hi, @toddaustin07 
`iot_bsp_wifi_get_scan_result` expects to get wifi ssid scan result from iot device (Raspberry-Pi in this case) perspective which would be used as base information to show available wifi list while device on-boarding at ST mobile application.
because `iot_bsp_wifi_get_scan_result` could be called during soft-ap mode, it should be implemented carefully.
some of BSPs/Chipsets can't support WiFi scan during soft-ap mode. the port layer should return cached - which is collected while station mode - wifi scan list as a fallback. 
I'm not sure the BCM wifi chipset could support wifi scanning during soft-ap mode. I think you can figure it out. Good luck !

Kwang-Hui, 2020-11-23:
> Wow great. Congratulations!
>
> I'll look into SDK code what you mentioned. Thanks again.

I also fixed the problem you reported. Please refer below commits. Thank you again.
 - https://github.com/SmartThingsCommunity/st-device-sdk-c/commit/07d4cd241bb607a2398bd1275240a1ed7b9a38f2
 - https://github.com/SmartThingsCommunity/st-device-sdk-c/commit/7b53919ebafb54a750c167b9b073ad2dc9758349
 - https://github.com/SmartThingsCommunity/st-device-sdk-c/commit/70de93676962f2536cc833db3875d30b82e6b3f6

toddaustin07, 2020-11-23:
Thanks for your reply!  I think I'm pretty clear on the scan_results function and I have that working fine.  And yes, I understand that if you are in AP mode, it needs to be implemented carefully.  But I'm still not clear what the purpose of a **scan mode** request is and what task it is supposed to perform.  Is it simply to take the device **out** of AP mode so that a scan_results can subsequently be called? 

And thank you for the fix to the open authorization bug!

toddaustin07, 2020-11-29:
Suggestion:

during st_conn_start, there are a number of warning messages displayed:

W [IOT]: http_packet_read(103) > Connection closed

I am seeing about 6 of these messages during onboarding.

My understanding is that HTTP 103 error is an informational status code to the client that the server is likely to send a final response.  Unless this is unexpected, I would suggest not displaying this particular message as a warning.

Kwang-Hui, 2021-01-04:
> Thanks for your reply! I think I'm pretty clear on the scan_results function and I have that working fine. And yes, I understand that if you are in AP mode, it needs to be implemented carefully. But [...]

`Scan mode` doesn't mean **out of AP mode**.
it is changing WiFi supplicant to support `Scanning` and it depends on H/W's capability.
Currently changing it to `Scan mode` at two points.
 - At the beginning of provisioning.
 - During provisioning.
(It can be changed in the future)

Mostly it isn't matter when `scan mode` is set at the beginning of provisioning.
But called at `during provisioning` . it shouldn't stop soft AP mode. but make it possible to WiFi scan. 
if your H/W doesn't support `wifi scan` during soft AP mode. you can return cached scan result as alternatives.

Kwang-Hui, 2021-01-04:
If you have further quires, Please open new issue.
Thanks

closed 2021-01-04 by Kwang-Hui