SmartThingsCommunity/st-device-sdk-c #65: HTTP 103 "Error" Warnings

toddaustin07, 2020-11-29:
Suggestion:

during st_conn_start, there are a number of warning messages displayed:

W [IOT]: http_packet_read(103) > Connection closed

I am seeing about 6 of these messages during onboarding.

My understanding is that HTTP 103 error is an informational status code to the client that the server is likely to send a final response. Unless this is unexpected, I would suggest not displaying this particular message as a warning.  Or perhaps as informational only, although I don't think it is of any particular value to users.

Kwang-Hui, 2020-11-30:
Hi @toddaustin07 ,

```
W [IOT]: http_packet_read(103) > Connection closed
```
Actually this isn't meaning of HTTP 103 error, it just print out line number of `http_packet_read()`.
And this log could be seen when the client closed connection. it's behavior could be differ for each platform and mobile app version.

thanks!

toddaustin07, 2020-11-30:
Thanks Kwang-Hui - Are you saying this message can be ignored then?  Since it is being displayed in orange text as a Warning, rather than as just information, it implies there may be a problem...

Kwang-Hui, 2020-11-30:
We prefer to use http keep-alive session. that warning could come when the client uses http closed connection.
but device on-boarding would be worked even with closed connection request. so you can ignore it.
let' me think about to change log level from warning to information.

Thanks.

junyoun-kim, 2023-02-15:
@toddaustin07 Do you have more questions about this thread? If not, it might be appreciated if you kindly close this issue.

closed 2023-02-16 by toddaustin07