SmartThingsCommunity/st-device-sdk-c #69: New License accept on mobile app breaks onboarding process

toddaustin07, 2020-12-18:
Starting today, a license acceptance pops up while onboarding a new device.  However at that moment, the mobile app is connected to the device because it's in the middle of the onboarding process, so the license acceptance cannot be completed because it seems to need internet access to complete.  There is no way to cancel this other than to shut down the mobile app.  The core SDK code is then left in an unrecoverable state.

Kwang-Hui, 2020-12-20:
Hi @toddaustin07 ,
Thanks for reporting issue.
I'll check that issue.

toddaustin07, 2020-12-29:
So sorry, I didn't see your question and now I know my SmartApp has been updated since I posted, so I can't tell you.  My phone auto-updates my apps, so it would have been whatever the current version was at the time of my original post.
I now have Version 1.6.57-463 and have not seen the issue again.

BTW - I posted a question (of more general interest) on the ST community board (How to Build Direct Connected Devices) regarding the use of custom capabilities.  Hope you or someone from Samsung can answer sometime.  It's very quiet on those forums!...

toddaustin07, 2021-01-23:
I got another legal notice in the middle of device onboarding a couple days ago.  This caused the process to hang and required a restart on the device, although this time I was able to check the approve box on the mobile app and then cancel the onboarding process.  My IOS mobile app version is 1.6.57-463

Kwang-Hui, 2021-06-23:
Please reopen this issue if you still have same problem. Thanks.

closed 2021-06-23 by Kwang-Hui