SmartThingsCommunity/st-device-sdk-c #71: Random provisioning hangup

toddaustin07, 2020-12-31:
Hello and Happy New Year!

I want to report an occasional error I get during onboarding devices.  After going through the onboarding process many many times, I've encountered this problem often enough that I think it is worth reporting, even though I cannot reproduce the problem every time.  It seems to be a problem with the communication between the mobile app and core SDK. 

After starting the provisioning process, the mobile app appears to successfully connect to the access point but then occasionally hangs at the same place, which is the highlighted line in the output below, containing "master secret generation success" and then nothing else until the core SDK eventually times out.  The mobile app hangs as well. When the core SDK finally times out, it displays the lines following "master secret generation success" as seen below ("New state changing timeout").

It seems the best way to clear this problem is to terminate and restart the mobile app and try again.

My mobile device is an iPhone6 with IOS version 12.4.8 and SmartThings App version 1.6.57-463.

Onboarding...connected to mobile
I [IoT]: http_msg_handler(385) > get cmd[0] ok
W [IoT]: http_packet_read(103) > Connection closed
I [IoT]: _iot_easysetup_gen_post_payload(258) > waiting.. response for [1]
**I [IoT]: _es_keyinfo_handler(553) > master secret generation success**
W [IoT]: _do_cmd_tout_check(967) > New state changing timeout
I [IoT]: _do_iot_main_command(450) > curr_main_cmd:10, curr_main_state:1/1
W [IoT]: _do_recovery(1349) > state changing fail for 1, curr_state :1
E [IoT]: _do_recovery(1407) > Failed process [1] on time, STOP
I [IoT]: _do_update_timeout(196) > Current timeout : 4294967295 for 1/-2
I [IoT]: _do_status_report(300) > Call usr status_cb with 2/3
Onboarding...FAILED

Kwang-Hui, 2021-01-03:
That's pretty interesting.

Could you check the time difference between below two lines?
```
I [IoT]: _iot_easysetup_gen_post_payload(258) > waiting.. response for [1]
I [IoT]: _es_keyinfo_handler(553) > master secret generation success
```

Your symptom says it takes too long time for generating master secret at your device (100 or 300 seconds timeout).
The possible check point is...
 - ecdh computing time 
 - some deadlock of IPC : `iot_easysetup_request` uses queue. if there is some blocking issue on queue. this could happen.

My testing result indicates ecdh computing is done within timeout even in ESP8266 (single core MCU).
So I'd like to suggest you IPC problem at first.
BTW, Could you let me know your test environment? (DUT's H/W, OS, etc..)

toddaustin07, 2021-01-04:
The display output I had copied and pasted into a file doesn't have timestamps so I can't tell you the time difference.

To be clear, the master secret generation success line is displayed, and THEN it hangs up.  So I thought that meant that the master secret had already been created.  It was hanging up on whatever was suppose to happen NEXT.

I'm testing on a Raspberry Pi 3b, with Raspbian Buster.  There's not much running at the time I'm doing the onboarding, so not sure how there could be a timeout due to CPU availability.  

I have not explored the logging options for the core SDK, and I've been meaning to ask you about this.  It looks like there is a debugging-level set of messages, but those are not displayed on the terminal.  Is there a way to turn those on?  Also, is there an option to log ALL messages to a file INSTEAD of displaying them on the screen?

I'll try to capture more info for you, for the next time it happens.

Kwang-Hui, 2021-01-04:
Thanks for update.
I think you can simply add **file write** and timestamp at `port/XXX/iot_bsp_debug_xxx.c`.
and turning `CONFIG_STDK_IOT_CORE_LOG_LEVEL_DEBUG` config would be helpful for debugging.

toddaustin07, 2021-02-20:
Hello - I've been able to capture a debug-level log for this issue.  This same hang occurs in about 25-30% of onboarding attempts.   See my prior posts for my configuration info.

Below I have two logs - the first one is when the failure occurs.  The second one is a successful onboarding process with no errors.  Comparing the two allows us to see exactly where the error is happening.  Both logs begin with the 'master secret generation success' log message.  There are no errors prior to this log message, but if you need to see that part of the log, I can post that too.

I see there are references to time/date.  Please consider our earlier conversation about the problems with setting the Raspberry Pi clock.  I had removed that function from the BSP system module.  Is it possible that this is a clock synchronization problem?  I don't understand your code enough to know what it is doing when the state changing timeout occurs, but it appears to have something to do with key generation and exchange with the mobile app.

I hope these logs can help you understand where the problem is.  Thank you.

=======================================================================================
                                                   FAILED LOG
=======================================================================================
I [IoT]: _es_keyinfo_handler(553) > master secret generation success
D [IoT]: iot_security_base64_decode_urlsafe(221) > src: 36@0x74701468, dst: 28@0x74701f08
D [IoT]: iot_security_base64_decode_urlsafe(252) > done: 25@0x74701f08
D [IoT]: _es_keyinfo_handler(584) > datetime = 2021-02-20T17.58.03 +0000
D [IoT]: iot_security_base64_decode_urlsafe(221) > src: 36@0x74700e00, dst: 28@0x74701f08
D [IoT]: iot_security_base64_decode_urlsafe(252) > done: 25@0x74701f08
D [IoT]: _es_keyinfo_handler(620) > regionaldatetime = 2021-02-20T11.58.03 -0600
D [IoT]: iot_security_base64_decode_urlsafe(221) > src: 20@0x747012b0, dst: 16@0x74700f38
D [IoT]: iot_security_base64_decode_urlsafe(252) > done: 15@0x74700f38
D [IoT]: _es_keyinfo_handler(653) > timezoneid = America/Chicago
D [IoT]: iot_security_cipher_aes_encrypt(370) > input = 26@0x74701d90
D [IoT]: _iot_security_be_software_cipher_aes(1060) > input:  26@0x74701d90
D [IoT]: _iot_security_be_software_cipher_aes(1061) > key:    32@0x74701710
D [IoT]: _iot_security_be_software_cipher_aes(1062) > iv:     16@0x74700dc8
D [IoT]: iot_security_cipher_get_align_size(278) > data size = 26, type = 3
D [IoT]: iot_security_cipher_get_align_size(318) > align size = 32
D [IoT]: _iot_security_be_software_cipher_aes(1127) > key:    32@0x74701710
D [IoT]: iot_security_cipher_aes_encrypt(384) > output = 32@0x74701ae0
D [IoT]: iot_security_base64_encode_urlsafe(178) > src: 32@0x74701ae0, dst: 45@0x74702168
D [IoT]: iot_security_base64_encode_urlsafe(192) > done: 44@0x74702168
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
D [IoT]: iot_os_eventgroup_set_bits(320) > write_size = 3
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
D [IoT]: iot_os_eventgroup_wait_bits(291) > read_size = 3
W [IoT]: _do_cmd_tout_check(968) > New state changing timeout
D [IoT]: iot_os_eventgroup_set_bits(320) > write_size = 3
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
D [IoT]: iot_os_eventgroup_wait_bits(291) > read_size = 3
D [IoT]: _iot_main_task(1058) > cmd: 10

I [IoT]: _do_iot_main_command(450) > curr_main_cmd:10, curr_main_state:1/1
W [IoT]: _do_recovery(1352) > state changing fail for 1, curr_state :1
E [IoT]: _do_recovery(1410) > Failed process [1] on time, STOP
D [IoT]: iot_os_eventgroup_set_bits(320) > write_size = 3
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
I [IoT]: _do_update_timeout(196) > Current timeout : 4294967295 for 1/-2
I [IoT]: _do_status_report(300) > Call usr status_cb with 2/3
Onboarding...FAILED
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
I [IoT]: _do_cmd_tout_check(981) > New state updated for -2
D [IoT]: iot_os_eventgroup_wait_bits(291) > read_size = 3
D [IoT]: _iot_main_task(1058) > cmd: 10

I [IoT]: _do_iot_main_command(450) > curr_main_cmd:10, curr_main_state:-2/-2
W [IoT]: _do_state_updating(1632) > Iot-core task will be stopped, needed ext-triggering

D [IoT]: iot_easysetup_deinit(500) > ENTER >>>>
I [IoT]: http_cleanup_all_connection(40) > close listen socket
I [IoT]: http_cleanup_all_connection(47) > close accept socket
I [IoT]: es_http_deinit(187) > http tcp deinit complete!
I [IoT]: iot_easysetup_deinit(524) > IOT_STATE_PROV_ES_DONE
I [IoT]: iot_bsp_wifi_set_mode(836) > [rpi] Requested mode OFF
I [IoT]: _SoftAPControl(1266) > [rpi] SoftAP service stop issued
I [IoT]: _do_status_report(300) > Call usr status_cb with 1/0
Idle, not connected...staying
D [IoT]: iot_os_eventgroup_set_bits(320) > write_size = 3
I [IoT]: _do_update_timeout(196) > Current timeout : 4294967295 for -1/-1
I [IoT]: _do_status_report(291) > Unsupported state -1 for 0/0
[... 650 lines cut ...]
Connected to MQTT server
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
D [IoT]: iot_os_eventgroup_wait_bits(291) > read_size = 3
D [IoT]: _iot_main_task(1058) > cmd: 0

I [IoT]: _do_iot_main_command(450) > curr_main_cmd:0, curr_main_state:7/8
I [IoT]: iot_cap_call_init_cb(1074) > Call init_cb for contactSensor capability
[... cut ...]
{"deviceEvents":[{"component":"main","capability":"partyvoice23922.zonebypass","attribute":"zoneBypass","value":"--","providerData":{"sequenceNumber":3,"timestamp":"1613844798555"}}]}
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
Sequence number return : 3
D [IoT]: iot_os_eventgroup_set_bits(320) > write_size = 3
D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x
D [IoT]: _iot_net_tls_write(320) > 231@0x74700c38
I [IoT]: _do_cmd_tout_check(981) > New state updated for 8

toddaustin07, 2021-03-25:
Hello - Do you have any new thoughts on this provisioning issue that I reported? 

I am now testing on a Raspberry Pi Zero W and this hang happens every time on this platform.   It looks to be a breakdown in communication between the mobile app and the core SDK during a security key exchange. 

'master secret generation success' is the last non-debug message that is displayed (line 147 in log file attached below) and then it hangs for a very long time (a couple minutes) before timing out (line 171) .  The mobile app never shows an error or times out.

I am now using an iPhone 12 with IOS version 14.3.  (I was hoping a new mobile phone might solve the issue, but no!) and SmartThings mobile app version 1.6.64-490

Now that I can recreate this problem every time, do you have any further recommendations on how to debug this so we can figure out the problem?

Here the latest message log:
[msg.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6201096/msg.log)

toddaustin07, 2021-03-28:
Hello again.  I have an additional log file that might be useful to help debug this issue.  This is a tcp dump of traffic captured during an attempted onboarding when the error occurs.
[dump3.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6216841/dump3.log)

I'm afraid interpreting this file is beyond my expertise, but it may be of use to someone there.

As mentioned above, I am not able to successfully onboard a device app and the error occurs in the same place every time: 'Step 1' seems to complete ok, but it appears there is no response from the mobile app to begin 'Step 2'.  

I've not gotten any responses from anyone on this problem lately, so  PLEASE RESPOND!!  I need your expertise to help figure out this problem, and now that I can reproduce it every time, we should be able find where things are hanging up between the mobile app and core SDK.

Kwang-Hui, 2021-03-29:
Hi @toddaustin07 ,

Sorry for late responding.
Unfortunately it is not easy for me to reproduce your issue on my side. 
The posix version is used only for unit test purpose on my side.

I don't believe there is any protocol level issues. I suspect performance or incompatible implementation of some posix porting layer.

To get more detailed information what you have trouble now. I posted two test commits here https://github.com/SmartThingsCommunity/st-device-sdk-c/pull/85

Could you please leave your device logs once again with adding above two commits?

Thanks.

Kwang-Hui Cho

toddaustin07, 2021-03-29:
Thank you for your reply.  I will add a time stamp log.  I have one in seconds but I think you probably need milliseconds precision.

I don’t understand what you want for adding ‘event group’ to the log.  Can you please clarify?  How do I do this?

Kwang-Hui, 2021-03-29:
I'm suspecting that it doesn't wakeup from the http response event at https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/7bad26224142381bf0e24b371e480d42bdcfa6b9/src/easysetup/http/iot_easysetup_http.c#L263

```
	for( ; ; ) {
		curr_event = iot_os_eventgroup_wait_bits(ctx->iot_events,
				IOT_EVENT_BIT_EASYSETUP_RESP, true, IOT_OS_MAX_DELAY);
		if (curr_event & IOT_EVENT_BIT_EASYSETUP_RESP)
			break;
	}
```

I hope that adding more information by https://github.com/SmartThingsCommunity/st-device-sdk-c/pull/85/commits/7bad26224142381bf0e24b371e480d42bdcfa6b9 could help to clarify the problem.

toddaustin07, 2021-03-29:
The Pi Zero is slower hardware than the Pi model 4 so I wonder if it is a timing issue.  I  see this same hang occasionally on the Pi model 4 but on Pi Zero it occurs EVERY time.  Both are running same version of Raspberry Pi OS (based on Debian buster).

Further info for you:  I’m working with someone else who is having same problem using Android phone (mine is iPhone), so problem doesn’t appear dependent on mobile app OS version.

Kwang-Hui, 2021-03-29:
We have 5 min timeout, But that is sufficient time even on the MCU level device - which is much slower than Pi zero.
And as you described above there are unnecessary 4 min waiting.

So I suspect that the incomplete port of eventgroup API makes this trouble.

toddaustin07, 2021-03-29:
Hello - Here is the new log file with timestamp and eventgroup info:

[newdebug.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6224214/newdebug.log)

Kwang-Hui, 2021-03-30:
Hi @toddaustin07 ,
I could see something strange from your log.

```
03/29/21 14:46:31.494 D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x08x (8)
```
As you can see in the code here. the `EVENT_MAX` is `8`. So the `i` shouldn't be `8`. I'm not sure why this happen.
https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/7bad26224142381bf0e24b371e480d42bdcfa6b9/src/port/os/posix/iot_os_util_posix.c#L313

And one more thing, actually I updated the log printing part like below to add missing `%` (fixes `0x08x` --> `0x%08x`) please update your code also.
```
@@ -312,12 +312,12 @@ int iot_os_eventgroup_set_bits(iot_os_eventgroup* eventgroup_handle,

        for (int i = 0; i < EVENT_MAX; i++) {
         if (eventgroup->group[i].id == (eventgroup->group[i].id & eventgroup->event_status)) {
-            IOT_DEBUG("already set 0x08x", eventgroup->group[i].id);
+            IOT_DEBUG("already set 0x%08x (%d)", eventgroup->group[i].id, i);
             continue;
         }

```

toddaustin07, 2021-03-30:
Here is the updated log:
[newdebug2.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6225501/newdebug2.log)

I see what you mean about i = EVENT_MAX.  That is indeed strange. Could the continue statement be causing this?  You'll see in this new log that i < EVENT_MAX so it is not always happening

Kwang-Hui, 2021-03-30:
I could see understand the few things.

1. `i = EVENT_MAX` 
Actually this wasn't happen. the log print has wrong because we missed `0x%08x`. The `eventgroup->group[i].id` has printed at `(%d)` instead of `i`
```
IOT_DEBUG("already set 0x08x (%d)", eventgroup->group[i].id, i);
```

2.  The EasySetup flow goes like below.
`/deviceinfo` --> `/keyinfo` --> `/confirminfo` --> ...
But just like you mention with TCP log, `iot-core` couldn't respond for `/keyinfo` because
the `iot_os_eventgroup_set_bits` for the `IOT_EVENT_BIT_EASYSETUP_RESP` doesn't wake up.

And this log is alos very strange. because I can't find to set bits again for IOT_EVENT_BIT_EASYSETUP_RESP after `iot_os_eventgroup_wait_bits` with `clear_on_exit` as true.
```
03/29/21 20:13:27.955 D [IoT]: iot_os_eventgroup_set_bits(315) > already set 0x00000008 (3)
```

Kwang-Hui, 2021-03-30:
I'd like to suggest you to trace the behavior of `iot_os_eventgroup_wait_bits()` for IOT_EVENT_BIT_EASYSETUP_RESP on your test device. basically the posix port uses `select()` method to `pipe` handle.

toddaustin07, 2021-03-30:
select() function can be tricky to use; I have some experience with that 🙂 so I will look closely at that and try to have more tracing info for you when you get in tomorrow.

I very much appreciate your help with this!

toddaustin07, 2021-03-30:
Good morning - I have a new log file for you and some thoughts:

1) I saw that _iot_easysetup_gen_post_payload was calling wait_bits with timeout of max IOT time value - an unsigned int.  However this number is too big for Linux time_t which is implemented with long int.  I thought maybe the millisecond conversion for the timeval structure was corrupting the stack.   I put in some code to check for it and pass NULL to select in that case, since that is the proper way to wait with block.  Unfortunately this didn't solve our problem, but probably should be addressed in the code nevertheless.

2) Another possible issue is with fd_max list so I've added some log messages to display that info to make sure the select list is including the right fd for response.

3) I was concerned about the effects of multi-threading since the wait_bits function updates the global variable eventgroup->event_status in the case of clear_on_exit, so I added a mutex to see if that might fix the problem with the response bit being already set.  

4) I tried setting the timeout value parameter from iot_easysetup_gen_post_payload to 5 seconds instead of block to force a re-invocation of wait_bits and this appears to have gotten the processing a step further as you'll see in the log.  However then there was an Invalid command sequence error and an HTTP bad request error.

I'm still trying different things, but wanted to get this log posted for you to see in case you have any ideas.

[newdebug5.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6232752/newdebug5.log)

toddaustin07, 2021-03-30:
Another thought occurred to me:  when I have used select() myself in other projects, I know that you have to re-initialize the fd list each time you call it because when it returns, it modifies the list you passed into it.  I don't know how thread-safe select() is.  Since the wait_bits routine appears to be called by multiple threads & functions, I am wondering if our resp fd is being removed from another thread or function invoking select() prior to being called by _iot_easysetup_gen_post_payload.  

If this were the case then code would need to be added to restore all 'un-processed' fd's after each select() call.  This is what I have had to do in my own code when using select().

toddaustin07, 2021-03-31:
If wait_bits times out for a RESP bit call, it returns eventgroup->event_status, which seems to have the bit still set.  So event_status bit seems to not have gotten cleared from prior step.

There seem to be multiple functions setting the RESP bit.  In the logs you can see the 'already set' messages, even though the last time it was cleared.  That is strange.  

I am having a hard time understanding the purpose of eventgroup->group[i].id vs. eventgroup->event_status.  It's not clear to me how and where these two values are set and reset, but it seems to be a synchronization problem between them.

toddaustin07, 2021-03-31:
Here is a log where I dump out eventgroup at every entry and exit of wait_bits and set_bits.  The multi-threading makes it challenging to follow in places.   You'll also have to scroll through all the wait_bits calls that timeout, but at least it's complete!

[newdebug9.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6233541/newdebug9.log)

Kwang-Hui, 2021-03-31:
Hi @toddaustin07 
Thanks for your investigation.
I'll also look into the log you attacked.
It would takes some time.

toddaustin07, 2021-03-31:
OK thanks.  I think I'm getting closer to the problem.

There are synchronization issues between the set and wait functions around the event_status value.  There also seems to be a problem where _iot_easysetup_gen_post_payload is expecting event_status to be set with RESP bit upon return from wait_bits call, but the wait_bits function doesn't ever update event_status with the bits value.  (It only the NOT value to clear).  

Actions I'm trying:
1)  protect all of set_bits execution with a mutex
2) add eventgroup->eventstatus |= bits;  to end of wait_bits function

Will keep you posted.

toddaustin07, 2021-03-31:
Last log for today:  I made progress getting to the next step but am running into a an invalid command sequence error and HTML errors, so something else must be broken still.  Here's my last log for today!...

I've cut down on the log messages so you don't have to scroll through all the select() timeout cases.

[newdebug15.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6234049/newdebug15.log)

Kwang-Hui, 2021-03-31:
> OK thanks. I think I'm getting closer to the problem.
> 
> There are synchronization issues between the set and wait functions around the event_status value. There also seems to be a problem where _iot_easysetup_gen_post_payload is expecting event_status to be set with RESP bit upon return from wait_bits call, but the wait_bits function doesn't ever update event_status with the bits value. (It only uses the NOT value to clear).
> 
> Actions I'm trying:
> 
> 1. protect all of set_bits execution with a mutex
> 2. add eventgroup->eventstatus |= bits;  to end of wait_bits function
>    -- OR--
>    changing _iot_easysetup_gen_post_payload logic where it calls wait_bits to match that used in the _iot_easysetup_gen_get_payload, which doesn't use a for loop and doesn't check RESP bit upon return from wait_bits.  (WHY DO THESE FUNCTIONS USE DIFFERENT LOGIC TO CALL WAIT_BITS??!)
> 
> Will keep you posted.

I agree with your opinion. this log indicates there is some synchronization issue between `posix` version of `eventgroup` apis.
let me think how to resolve this.

Kwang-Hui, 2021-03-31:
Could you try with this https://github.com/SmartThingsCommunity/st-device-sdk-c/pull/85/commits/25497ab73ba2277bb8166f323150c044a59bc5c4 commit?

garrettpatterson, 2021-03-31:
Built the posix-debug branch: still appears to hang/not respond to the /keyinfo request, output log attached.

Reviewing the TCP dump - the connection appears to continue to be active with TCP keep-alive pings between the Pis AP and mobile device (I'm on an Android pixel 4a) until it resets itself back to wifi.

[st-device-sdk-c-example_log_202103311122.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6238952/st-device-sdk-c-example_log_202103311122.txt)

Kwang-Hui, 2021-03-31:
Hi @garrettpatterson @toddaustin07 ,
Thanks for your updates.

The log from @garrettpatterson still shows eventgroup_wait_bits doesn't respond properly.

But the log from @toddaustin07 is different. eventgroup_wait_bits has worked. But it fails with other reasons.
The iOS version of ST app still uses connection-closed method, so the socket should be closed after sending respond of `/keyinfo`  from device just like below logs (captured from `mutexfix1.log`)
```
03/31/21 13:27:17.563 W [IoT]: http_packet_read(103) > peer connection closed
03/31/21 13:27:17.563 I [IoT]: http_cleanup_accepted_connection(59) > close accept socket
```

@toddaustin07 , Could you add addtional debug logs like below?
```
--- a/src/easysetup/http/tcp/iot_easysetup_httpd.c
+++ b/src/easysetup/http/tcp/iot_easysetup_httpd.c
@@ -83,6 +83,7 @@ static int process_accepted_connection(HTTP_CONN_H *handle)
                        }
                        payload = rx_buffer + http_request_header_len;
                }
+               IOT_DEBUG("%s", rx_buffer);
 
                if(err != IOT_ERROR_NONE) {
                        http_msg_handler(cmd, &tx_buffer, D2D_ERROR, payload);
@@ -108,6 +109,7 @@ static int process_accepted_connection(HTTP_CONN_H *handle)
                        IOT_ES_DUMP(IOT_DEBUG_LEVEL_ERROR, IOT_DUMP_EASYSETUP_SOCKET_SEND_FAIL, errno);
                        return IOT_ERROR_EASYSETUP_INTERNAL_SERVER_ERROR;
                }
+               IOT_DEBUG("http_packet_send result: %d", len);
        }
 }

```

Kwang-Hui, 2021-04-01:
I updated code diff for logging

```
--- a/src/easysetup/http/tcp/iot_easysetup_httpd.c
+++ b/src/easysetup/http/tcp/iot_easysetup_httpd.c
@@ -24,6 +24,7 @@
 #ifdef CONFIG_STDK_IOT_CORE_EASYSETUP_HTTP_USE_SOCKET_API
 #include "iot_easysetup_http_socket.h"
 #endif
+#include "iot_util.h"
 
 #define RX_BUFFER_MAX    1024
 
@@ -68,7 +69,7 @@ static int process_accepted_connection(HTTP_CONN_H *handle)
                        }
                        return err;
                }
-
+               IOT_DEBUG(">>> %s", rx_buffer);
                content_len = 0;
                err = es_msg_parser(rx_buffer, sizeof(rx_buffer), &payload, &cmd, &type, &content_len);
 
@@ -84,6 +85,8 @@ static int process_accepted_connection(HTTP_CONN_H *handle)
                        payload = rx_buffer + http_request_header_len;
                }
 
+               iot_util_dump_mem("dump", (uint8_t*) payload, content_len);
+
                if(err != IOT_ERROR_NONE) {
                        http_msg_handler(cmd, &tx_buffer, D2D_ERROR, payload);
                }
@@ -108,6 +111,7 @@ static int process_accepted_connection(HTTP_CONN_H *handle)
                        IOT_ES_DUMP(IOT_DEBUG_LEVEL_ERROR, IOT_DUMP_EASYSETUP_SOCKET_SEND_FAIL, errno);
                        return IOT_ERROR_EASYSETUP_INTERNAL_SERVER_ERROR;
                }
+               IOT_DEBUG("http_packet_send result: %d", len);
        }
 }

```

toddaustin07, 2021-04-01:
Here is the new log with your requested debug statements added:
[mutexfix2.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6240186/mutexfix2.log)

Strange that there are two /keyinfo sent, but is that really the case?  Are we seeing another synchronization issue?

Kwang-Hui, 2021-04-01:
I think the `mutexfix1.log` , `mutext2.log` shows that evetgroup problem has fixed.

The multiple `/keyinfo` is different story.
log sniff from `mutexfix1.log`  says `/keyinfo` is called approx. 3 seconds period.
```
03/31/21 13:27:17.571 D [IoT]: es_msg_parser(83) > method=POST, uri=/keyinfo
...
03/31/21 13:27:20.901 D [IoT]: es_msg_parser(83) > method=POST, uri=/keyinfo
...
03/31/21 13:27:24.013 D [IoT]: es_msg_parser(83) > method=POST, uri=/keyinfo
```
Currently the iOS application repeats same http api again if the device doesn't send response packet within 3 seconds.

We could suspect that the response packet is still buffered in the device inside. (185 bytes isn't big enough to filling tcp buffer)
Actually to avoid this kind of buffering problem SDK code set `TCP_NODELAY` option
https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/00921d860059a6563053838a22f2c452cc72ccfc/src/easysetup/http/tcp/iot_easysetup_http_socket.c#L196

If you can analysis TCP dump like previous, we can make it clear.

toddaustin07, 2021-04-01:
Garrett had a better TCP dump utility than mine - at least it seemed to show more readable output.  Let me try running the one I have, but I'll ask Garrett what he's using.

toddaustin07, 2021-04-01:
Here's my TCP dump:

[newtcp.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6240751/newtcp.log)

toddaustin07, 2021-04-01:
Sorry about the red herring in my prior post, I was partially fooled by printing payload before the second receive.  It appears from my testing that sometimes that /keyinfo message is sent in one chunk and sometimes in two chunks.

However it still concerns me that in the TCP log I posted earlier today, it does show that the second chunk of the /keyinfo message was never sent in one case.  That may be a whole other TCP-level reliability problem, because as you'll see in my updated logs below, that was not a problem in this new test.  But the eventual resend of /keyinfo still occurred.

So now I'm back to thinking that maybe something is wrong with the /keyinfo response message, since the TCP log again shows a strange 'double ACK' for that message by the mobile device, and then mobile device proceeds to resend the /keyinfo message.

Here are current updates to both the TCP and core message logs (they are from the same test):
[mutexfix3.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6241281/mutexfix3.log)
[tcpdump3.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6241282/tcpdump3.log)

Kwang-Hui, 2021-04-01:
> Sorry about the red herring in my prior post, I was partially fooled by printing payload before the second receive. It appears from my testing that sometimes that /keyinfo message is sent in one [...]

Thanks for sending update.
If you don't mind, Could you send me the whole pcap with `tcpdump -w test.pcap` from Rasp. pi W?

My observation about iOS's POST behavior was sending two chunks separated by HTTP header and body part.

I also could see something strange from `tcpdata.log`. 
The header chunk for `/keyinfo` is exist. but can't find the body part. but the device sent "200 OK" response with proper answer.
I think something missed at the tcpdump at that time.

I'll look at newly attached logs and get back to you. (mutexfix3.log, tcpdump3.log)

toddaustin07, 2021-04-01:
Hi again - I thought I'd post a TCP log of a WORKING provisioning session on a Pi Model 4 so we can compare. 

You'll see in this one that there is no strange second ACK given by the mobile for the /keyinfo response; and things go on properly from there.

[tcpworking.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6241443/tcpworking.log)

toddaustin07, 2021-04-01:
Here is the the entire tcpdump file as you requested.  I let it run until the core timed out after 4 min.  I had to add the '.log' to the file name in order to be able to post it here...

[test.pcap.log](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6241512/test.pcap.log)

Kwang-Hui, 2021-04-01:
Thanks for the pcap log.

Could you send the iOS ST app log to us?
I'll send you video tutorial how to collect ST app log.

garrettpatterson, 2021-04-01:
So... excited to report after a fresh reboot on the PI, pull from the posix-debug branch and rebuild I had a successful device onboarding.  Log attached - after onboarding I'm having some device errors which I think are just related to my device config setup and what the example project was built as but it's making requests.  I'll be curious what @toddaustin07 finds.

When I stop the process and re-start it properly went back into device mode and making requests.

Reminder on my specs:
Raspberry Pi Zero W, full Raspbian Buster (10)
ST Mobile App in Developer mode on a Pixel 4a.

Couldn't get a TCPdump of this run.

[st_device_sdk_c_log_successful_20210401.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6245442/st_device_sdk_c_log_successful_20210401.txt)

toddaustin07, 2021-04-01:
Garrett- that's fantastic!  Did you happen to capture a log when you onboarded?  The one you attached seems to be post-onboarding.

toddaustin07, 2021-04-01:
Garrett - I know you may not want to mess with it now that it's working, but if you have a chance.....
Are you still running with sudo to get around the time-set issue?  If you are that means you are letting the core SDK set your clock.  I had put in a dummy return for that request in /src/port/bsp/posix/iot_bsp_system_posix.c and I am still running that way (without sudo).  I'll attach my replacement file below.  I'd be curious to see if this has any effect for you.  I will also re-enable that time check on my end to see if it makes a difference.

Reason I'm curious about this is that (if I understand things correctly) the master secret that gets sent back to mobile device after /keyinfo request I think gets generated partially based on the system clock.  Therefore if there is anything that the mobile app doesn't like about the key, that may be causing it to re-request.  Just a guess....

[iot_bsp_system_rpi.c.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6245734/iot_bsp_system_rpi.c.txt)

toddaustin07, 2021-04-01:
Kuang-Hui - Here is the mobile app log:
[SamsungConnect4iOS_Log.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6245827/SamsungConnect4iOS_Log.txt)

garrettpatterson, 2021-04-01:
Made the update that @toddaustin07 had suggested https://github.com/SmartThingsCommunity/st-device-sdk-c/issues/71#issuecomment-812091147 .  The settime still appears to be an issue but we are now getting HTTP responses sent back vs just appearing to "hang" after the /keyinfo.  

Should be more complete log output, and the HTTP response/requests (from 2 attempts to device onboard).

[settimeadjustlogs.pcap.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6246082/settimeadjustlogs.pcap.txt)
[run_log_202104011225.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6246083/run_log_202104011225.txt)

toddaustin07, 2021-04-01:
Hi Garrett - 

Regarding the timeset issue, do this and rebuild; it's still picking up the original file:
```
cd ~/st-device-sdk-c/build
rm stdk_iot_bsp_system_posix.o
```

After that you shouldn't have to use sudo since the timeset is skipped.

So I tried what you did: rebooted and did a fresh pull of the posix-debug branch.  Got more than just the mods for this particular issue:

src/deps/mbedtls/Makefile       
 src/easysetup/http/tls/iot_easysetup_http_tls.c 
 src/include/iot_capability.h                
 src/include/st_dev.h                         
 src/include/st_dev_version.h               
 src/iot_api.c                                  
 src/iot_capability.c                      
 src/port/bsp/posix/iot_bsp_debug_posix.c       
 src/port/net/mbedtls/iot_net_mbedtls.c       
 src/port/os/posix/iot_os_util_posix.c       
 src/port/os/tizenrt/iot_os_util_tizenrt.c     
 test/TC_FUNC_iot_wt.c                      
 tools/keygen/README.md               
 tools/keygen/stdk-keygen.py            

Unfortunately I continue to get the same error.  **Perhaps this narrows things down to an IOS issue!**

toddaustin07, 2021-04-03:
I found the problem!  I had originally copied a (unused) device_info.json file over from another Pi to use on the Pi Zero for my tests.  When I remembered this today, I tried creating a fresh serial number on the Zero, added it to Developer Workspace, and it onboarded without a problem!  I feel badly that I caused this problem and I apologize for the time wasted.  It does highlight the need for better error messages, however.  And I learned my lesson that serial numbers cannot be moved across computers.

At least we also fixed a real problem, which was the need for a mutex  in the iot_os_util_posix module.  Kwang-Hui, I assume that will get included in an upcoming release.

Thanks so much for all the assistance and again, I am sorry to be the cause of this last problem!

Kwang-Hui, 2021-04-04:
Good news!

Yes, the fix of posix  eventgroup would be part of next release.
I also realized that we need to update the log message of payload print what you mentioned at https://github.com/SmartThingsCommunity/st-device-sdk-c/issues/87.

Thanks

Kwang-Hui, 2021-04-04:
This issue is closing now. If you need further discussion, please open new issue again.

Thanks.

closed 2021-04-04 by Kwang-Hui