SmartThingsCommunity/st-device-sdk-c #77: Handling of wifi_init failures

toddaustin07, 2021-02-06:
Hello - I've observed that even if iot_bsp_wifi_init returns an error, the core SDK seems to ignore and continues anyway.  If an unrecoverable error is encountered during wifi init (device is not functioning or there is some configuration problem), are you expecting the wifi module to exit?  I would think the core SDK should have a chance to clean up and exit, or return the st_conn_start API call back to the application with an error.

stdk-scm, 2021-02-10:
Dear toddaustin07

Thanks for your advice to add error-handling during wifi init.
We need to add error-handling codes for iot_bsp_wifi_init()

Thanks

Kwang-Hui, 2021-03-10:
`iot_bsp_wifi_init()` error handling patch has been merged https://github.com/SmartThingsCommunity/st-device-sdk-c/commit/b1fd10dfd64ad50536eab147e1e28a7f57877166#diff-a424814a77d283b256630dba916da0315f13c30f82e9cdcefb477278651f77e3

closed 2021-03-10 by Kwang-Hui