SmartThingsCommunity/st-device-sdk-c #83: mbedtls reference issues during make

toddaustin07, 2021-03-22:
Hello - I've pulled a fresh copy of the sdk and have built the core library and ran make on the example app, but there is a new linker error appearing, apparently related to the recent mbedtls update:

`undefined reference to 'mbedtls_platform_zeroize'  in library ctr-drbg.c`

It looks like there is some unwanted coded being pulled into the make process - probably during the core sdk build.  
Here is a file with the specific messages:
[makeoutput.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c/files/6178890/makeoutput.txt)

There is also a minor warning error I'm now getting during the core sdk build, also related to the new mbedtls: 

`mbedtls/library/entropy_poll.c:24: warning: "_GNU_SOURCE" redefined #define _GNU_SOURCE`
`<command-line>: note: this is the location of the previous definition`

I don't know if the two things are related...

Kwang-Hui, 2021-03-23:
Hi @toddaustin07 ,

When I try build simply like below
```
git clone https://github.com/SmartThingsCommunity/st-device-sdk-c.git
cd st-device-sdk
make
```
it doesn't make build error (but has _GNU_SOURCE redefine warning).

I also try to swith it with `develop` branch. it' same.

Could you try again after `make clean` ?

or

Please check whether the mbedtls is up to date.
`mbedtls` submodule should be synchronized with `mbedtls @ fb1972d` ?
https://github.com/SmartThingsCommunity/st-device-sdk-c/tree/master/src/deps/mbedtls

I think you can use `git submodule status --recursive`.

toddaustin07, 2021-03-23:
Hi Kwang-Hui - You won't see the compiler error **until you make the example app** because the linker can't resolve the mbedtls_platform_zeroize reference.  When I make the core library, there is only the same warning you saw. 

mbedtls version is 2.16.0

It's the new code, since that function used to be 'mbedtls_zeroize' and now it is 'mbedtls_platform_zeroize'

If you still can't recreate the error when you make the example app, I will try to reload everything to a fresh install.

Kwang-Hui, 2021-03-23:
Hi @toddaustin07 ,

Oh I got your point. let me check further.

Kwang-Hui, 2021-03-25:
Hi @toddaustin07 ,
This commit (https://github.com/SmartThingsCommunity/st-device-sdk-c/commit/f212fea88654f1416afed42e6a0fe6f65e8eccd2) fixes the build break.
Please note this commit located at `develop` branch for now and will be merged to `mastser` branch soon.

Thanks.

closed 2021-03-25 by Kwang-Hui