SmartThingsCommunity/st-device-sdk-c #94: Error When Building Core SDK on Raspberry Pi

chrislope, 2021-06-17:
I got the following error when building the core SDK:

**In file included from src/easysetup/http/tcp/iot_easysetup_httpd.c:23:
src/include/http/iot_easysetup_http.h:22:10: fatal error: iot_easysetup_http_impl.h: No such file or directory
 #include "iot_easysetup_http_impl.h"**

I was able to work around it by updating the path in iot_easysetup_http.h to '''/src/port/http/socket/iot_easysetup_http_impl.h'''
```
Building SmartThings Core SDK

make[1]: Entering directory '/home/pi/st-device-sdk-c/src/deps/json'
  CC    cJSON/cJSON_Utils.o
  CC    cJSON/test.o
  CC    cJSON/cJSON.o
make[1]: Leaving directory '/home/pi/st-device-sdk-c/src/deps/json'
make[1]: Entering directory '/home/pi/st-device-sdk-c/src/deps/libsodium'
  CC    libsodium/src/libsodium/crypto_auth/hmacsha256/auth_hmacsha256.o
  CC    libsodium/src/libsodium/crypto_auth/crypto_auth.o
  CC    libsodium/src/libsodium/crypto_auth/hmacsha512/auth_hmacsha512.o
  CC    libsodium/src/libsodium/crypto_auth/hmacsha512256/auth_hmacsha512256.o
  CC    libsodium/src/libsodium/crypto_onetimeauth/poly1305/onetimeauth_poly1305.o
  CC    libsodium/src/libsodium/crypto_onetimeauth/poly1305/sse2/poly1305_sse2.o
  CC    libsodium/src/libsodium/crypto_onetimeauth/poly1305/donna/poly1305_donna.o
  CC    libsodium/src/libsodium/crypto_onetimeauth/crypto_onetimeauth.o
  CC    libsodium/src/libsodium/crypto_box/crypto_box.o
  CC    libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.o
  CC    libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.o
  CC    libsodium/src/libsodium/crypto_box/crypto_box_easy.o
  CC    libsodium/src/libsodium/crypto_box/crypto_box_seal.o
  CC    libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.o
  CC    libsodium/src/libsodium/crypto_shorthash/siphash24/ref/shorthash_siphashx24_ref.o
  CC    libsodium/src/libsodium/crypto_shorthash/siphash24/ref/shorthash_siphash24_ref.o
  CC    libsodium/src/libsodium/crypto_shorthash/siphash24/shorthash_siphash24.o
  CC    libsodium/src/libsodium/crypto_shorthash/siphash24/shorthash_siphashx24.o
  CC    libsodium/src/libsodium/crypto_shorthash/crypto_shorthash.o
  CC    libsodium/src/libsodium/crypto_hash/crypto_hash.o
  CC    libsodium/src/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.o
  CC    libsodium/src/libsodium/crypto_hash/sha256/hash_sha256.o
  CC    libsodium/src/libsodium/crypto_hash/sha512/hash_sha512.o
  CC    libsodium/src/libsodium/crypto_hash/sha512/cp/hash_sha512_cp.o
  CC    libsodium/src/libsodium/randombytes/sysrandom/randombytes_sysrandom.o
  CC    libsodium/src/libsodium/randombytes/randombytes.o
  CC    libsodium/src/libsodium/randombytes/internal/randombytes_internal_random.o
  CC    libsodium/src/libsodium/crypto_secretbox/crypto_secretbox_easy.o
  CC    libsodium/src/libsodium/crypto_secretbox/xchacha20poly1305/secretbox_xchacha20poly1305.o
  CC    libsodium/src/libsodium/crypto_secretbox/crypto_secretbox.o
  CC    libsodium/src/libsodium/crypto_secretbox/xsalsa20poly1305/secretbox_xsalsa20poly1305.o
  CC    libsodium/src/libsodium/crypto_stream/crypto_stream.o
  CC    libsodium/src/libsodium/crypto_stream/xchacha20/stream_xchacha20.o
  CC    libsodium/src/libsodium/crypto_stream/chacha20/dolbeau/chacha20_dolbeau-ssse3.o
  CC    libsodium/src/libsodium/crypto_stream/chacha20/dolbeau/chacha20_dolbeau-avx2.o
  CC    libsodium/src/libsodium/crypto_stream/chacha20/ref/chacha20_ref.o
  CC    libsodium/src/libsodium/crypto_stream/chacha20/stream_chacha20.o
  CC    libsodium/src/libsodium/crypto_stream/salsa2012/stream_salsa2012.o
  CC    libsodium/src/libsodium/crypto_stream/salsa2012/ref/stream_salsa2012_ref.o
  CC    libsodium/src/libsodium/crypto_stream/salsa20/xmm6/salsa20_xmm6.o
  CC    libsodium/src/libsodium/crypto_stream/salsa20/xmm6int/salsa20_xmm6int-sse2.o
  CC    libsodium/src/libsodium/crypto_stream/salsa20/xmm6int/salsa20_xmm6int-avx2.o
  CC    libsodium/src/libsodium/crypto_stream/salsa20/ref/salsa20_ref.o
  CC    libsodium/src/libsodium/crypto_stream/salsa20/stream_salsa20.o
  CC    libsodium/src/libsodium/crypto_stream/xsalsa20/stream_xsalsa20.o
  CC    libsodium/src/libsodium/crypto_stream/salsa208/stream_salsa208.o
  CC    libsodium/src/libsodium/crypto_stream/salsa208/ref/stream_salsa208_ref.o
  CC    libsodium/src/libsodium/crypto_pwhash/crypto_pwhash.o
  CC    libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/pbkdf2-sha256.o
  CC    libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/crypto_scrypt-common.o
  CC    libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.o
  CC    libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/pwhash_scryptsalsa208sha256.o
  CC    libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/sse/pwhash_scryptsalsa208sha256_sse.o
  CC    libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/scrypt_platform.o
[... 151 lines cut ...]
  CC    src/deps/cbor/tinycbor/src/cborparser_dup_string.o
  CC    src/deps/cbor/tinycbor/src/cborencoder.o
  CC    src/deps/cbor/tinycbor/src/cborerrorstrings.o
  CC    src/deps/cbor/tinycbor/src/cbortojson.o
  CC    src/deps/cbor/tinycbor/src/cborencoder_close_container_checked.o
  CC    src/port/bsp/posix/iot_bsp_wifi_rpi.o
  CC    src/port/bsp/posix/iot_bsp_random_posix.o
  CC    src/port/bsp/posix/iot_bsp_nv_data_posix.o
  CC    src/port/bsp/posix/iot_bsp_system_rpi.o
  CC    src/port/bsp/posix/iot_bsp_debug_posix.o
  CC    src/port/bsp/posix/iot_bsp_fs_posix.o
  CC    src/port/os/posix/iot_os_util_posix.o
  CC    src/port/net/mbedtls/iot_net_mbedtls.o
  CC    src/easysetup/iot_easysetup_st_mqtt.o
  CC    src/easysetup/http/iot_easysetup_http_parser.o
  CC    src/easysetup/http/iot_easysetup_http.o
  CC    src/easysetup/http/tcp/iot_easysetup_d2d_tcp.o
  CC    src/easysetup/http/tcp/iot_easysetup_httpd.o
In file included from src/easysetup/http/tcp/iot_easysetup_httpd.c:23:
src/include/http/iot_easysetup_http.h:22:10: fatal error: iot_easysetup_http_impl.h: No such file or directory
 #include "iot_easysetup_http_impl.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [/home/pi/st-device-sdk-c/make/common.mk:7: src/easysetup/http/tcp/iot_easysetup_httpd.o] Error 1
```

Kwang-Hui, 2021-06-17:
Hi @chrislope ,
Thanks for submitting issue.
I think you're using Make (Not CMake) for your testing.
So I tried clean build with just "make" command with commit 
"4367e87 Merge pull request #93 from SmartThingsCommunity/master_rel_1_6_2"
But the build error isn't reproduce at my side.

As you can see in the Makefile. That is related with `STDK_IOT_CORE_EASYSETUP_HTTP_USE_SOCKET_API`
```
ifneq ($(findstring STDK_IOT_CORE_EASYSETUP_HTTP_USE_SOCKET_API, $(STDK_CONFIGS)),)
HTTP_DIR = src/port/http/socket
endif
```

Please check your stdkconfig file has `STDK_IOT_CORE_EASYSETUP_HTTP_USE_SOCKET_API` like below? (https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/4367e87df5538aa46bf2cd5505abb30df80fbe91/stdkconfig#L9)
```
STDK_CONFIGS += STDK_IOT_CORE_EASYSETUP_HTTP_USE_SOCKET_API
```

chrislope, 2021-06-22:
Hi, @Kwang-Hui!  Thank you for looking into this issue.  Apparently the Makefile that I was using was modified and was missing the section you pasted above.  Please go ahead and close out this issue as a "dummy at the keyboard" error, haha.

closed 2021-06-22 by Kwang-Hui