SmartThingsCommunity/st-device-sdk-c #95: Change in QR Code scan procedure in new mobile app

toddaustin07, 2021-06-24:
Hello, did you miss me?!

I've noticed a change in the way the new mobile app behaves for developer mode devices.  We can no longer use the 'Scan QR code' button in the lower left of the Add device screen because when it scans the QR code generated by the qrgen tool, it says it is not a QR code for SmartThings devices.   Instead,  you always have to scroll through the device type icons to the developer test icon (displayed at bottom), select that, and then after a few of the onboarding screens you reach a QR code scan that DOES work.   I'm not sure if this behavior is by design or not.

Please note that above comments are based on the iOS version of the new mobile app.

Kwang-Hui, 2021-06-24:
Hi, @toddaustin07 

Yes, I missed you :-)
Could you share your test device's QR code?

Thanks.

toddaustin07, 2021-06-24:
[image]

Kwang-Hui, 2021-06-24:
Thanks @toddaustin07 .
Could you send me the ST app log about this issue?
The team has tried to reproduce this issue in our side but failed.
Your log file would be helpful to analysis this issue.

toddaustin07, 2021-06-24:
Remind me how to do that please (with new mobile app)?

Kwang-Hui, 2021-06-24:
(Right bottom conner of ST app) Menu --> Setting --> Help --> Report a problem --> Save to Files --> "Save file" --> "Go back if brwoser is comming" --> Home screen --> Files --> Send the file on email or attach to github issue

Thanks!

toddaustin07, 2021-06-24:
Please disregard this issue; the reason the QR code was being rejected was because I didn't not have developer mode enabled in the new app yet.  Evidently the app is smart enough to know when the QR code is for a developer mode/test device and rejects it when developer mode is not on.  That make sense!

closed 2021-06-24 by toddaustin07