SmartThingsCommunity/st-device-sdk-c #97: Not able to transfer data greater than 4kb from one device to another via MQTT

pramod-begur, 2021-07-14:
I have connected two devices via the MQTT and trying to transfer a file data of size > 4kb but at the subscriber end getting an error as _iot_net_tls_read(430) > mbedtls_ssl_read = -0x7200.

I could see the log that MQTT broker is able to publish the data to the subscriber but at subscriber error is thrown and disconnected.

So please help me to resolve this issue.

Kwang-Hui, 2021-07-14:
Hi @pramod-begur ,

Thanks for submitting issue.
Could you let me know which capability do you use for sending data and attach full device log for better understanding your symptom?

pramod-begur, 2021-07-15:
> Hi @pramod-begur ,
> 
> Thanks for submitting issue.
> Could you let me know which capability do you use for sending data and attach full device log for better understanding your symptom?

Thanks for the reply.
We are using st-device-sdk-c MQTT protocol to connect the devices.
From one linux device trying to publish a message with payload of size >4kb to other linux device using the MQTT protocol.
So when the message is published, at the subscriber device getting "_iot_net_tls_read(430) > mbedtls_ssl_read = -0x7200" error. Also not getting much logs except the mentioned error.
When we send chunked data of size less than 4kb, the messages are received at the subscriber device.

Kwang-Hui, 2021-07-15:
Hi @pramod-begur ,

This project - *st-device-sdk-c* - is designed to communicate [SmartThings capabilities](https://smartthings.developer.samsung.com/docs/api-ref/capabilities.html) via MQTT protocol.
So I'm little bit confused the purpose of your query :)
Could you let me know your purpose of device communication with using `st-device-sdk` ?

Kwang-Hui, 2021-07-19:
This issue has been discussed at other communication channel.
So closed.

closed 2021-07-19 by Kwang-Hui