SmartThingsCommunity/st-device-sdk-c #101: onboarding_config.json

toddaustin07, 2021-08-20:
I recently re-downloaded the onboarding_config.json file from an existing dev workspace profile.   The onboarding name is "DSC Zone Device".  Previously, the onboarding_config file contained a deviceOnboardingId value of "DSC_Zone_Devi".  When I redownloaded the file today, however, that value was "DSC Zone Devi".  In other words it was missing the underscores in place of the spaces.  This of course caused the device provisioning process to fail since the mobile app is trying to connect to an SSD that includes the '_' characters.  

Somehow there is a code path in the Developer Workspace that is missing generating the underscores in place of spaces for the onboarding name in the onboarding_config.json file.

Kwang-Hui, 2021-08-22:
Hi @toddaustin07,
Thanks for the issue reporting. Sorry for the inconvenience.
The improvement of Developer workspace might impact to your developing project. I'll back with internal discussion with teams.

Kwang-Hui, 2021-08-23:
Hello @toddaustin07 ,
Could you kindly let me know when the project was created for better analysis?

Thanks.

toddaustin07, 2021-08-24:
Developer Workspace shows that the device profile was last edited December 29, 2020.

toddaustin07, 2021-09-01:
Any update on this?

Kwang-Hui, 2021-09-01:
Hi, @toddaustin07 , Sorry for the late update.
It seems side effect of enhancement of developer workspace. The teams are under investigating how to fix this issue.
Please edit downloaded "onboarding_config.json" to have deviceOnboardingId "DSC_Zone_Devi" as a workaround.
Thanks.

toddaustin07, 2021-09-02:
Thanks for the update Kwang-Hui - Yes, I've already manually edited the json so I could get the device onboarded.  But I'm worried about anyone else using my RPI package having this problem.   Is this an issue only due to using a profile created many months ago?  If one were to create a new profile today, would the onboarding json still have the wrong deviceOnboardingId?

Kwang-Hui, 2021-09-02:
If any one create new developer workspace project today, that project also has "DSC Zone Devi" form not "DSC_Zone_Devi".
But it doesn't make problem because the cloud instance also registered as "DSC Zone Devi".

While the old project got a failure due to mismatch of cloud registered old style "DSC_Zone_Devi" and new style "DSC Zone Devi" onboarding_config.json.

Thanks.

Kwang-Hui, 2021-09-08:
Hi @toddaustin07 ,
The server side fix has been deployed.
Could you re-download the onboarding_config.json file again?

Thanks.

toddaustin07, 2021-09-09:
I just did and the underscores are there now.

**THANKS** to you and the team for getting this fixed!

Kwang-Hui, 2021-09-10:
This issue is being closed.

closed 2021-09-10 by Kwang-Hui