SmartThingsCommunity/st-device-sdk-c #107: New wifi event callbacks in iot_bsp_wifi_xxx.c

toddaustin07, 2021-12-11:
You have added two new functions:
```
iot_error_t iot_bsp_wifi_register_event_cb(iot_bsp_wifi_event_cb_t cb)
void iot_bsp_wifi_clear_event_cb(void)
```

to the porting modules /src/port/bsp/xxx/iot_bsp_wifi_xxx.c

These were added with no corresponding documentation about their purpose, and cause Make undefined reference errors for ported platform modules since they are appear to be required.

This change was not mentioned in the v1.7.0 release notes.

Can you please post here more information on these two functions?  What is their purpose and what functions are they expected to perform?  

Please also update the [porting guide](https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/master/doc/porting_guide.md) to include these additional required functions.

In the future please document these kind of changes that could affect ports for other platforms.

toddaustin07, 2021-12-12:
There also appears to be a new make config option:
```
STDK_IOT_CORE_EASYSETUP_DISCOVERY_SSID
```

What is this for, and is it related to the callback function?  I've noted by looking at some of the included BSP platform files, the register and clear callback functions do nothing?...

toddaustin07, 2022-01-01:
Hello???!

jy0209-kim, 2022-01-06:
Dear,

First of all, I'm sorry for late reply and thank you for your report.
Those APIs are included for detecting station connection/disconnection during SoftAP mode.

```
iot_error_t iot_bsp_wifi_register_event_cb(iot_bsp_wifi_event_cb_t cb)
: This is to register callback called when the station connection/disconnection is detected.

void iot_bsp_wifi_clear_event_cb(void)
: This is to clear the callback
```

Thus, if the device support detecting of station connection/disconnection in SoftAP mode, the callback could be registered and used in core.
We found that there's issue in core SDK code when a device doesn't support the detection of station connection/disconnection in SoftAP mode thanks to you.
We'll update it, the code, soon and the porting guide.
Thank you for your concern again.

BR, Jiyoung.

toddaustin07, 2022-01-07:
Thank you for your reply.  Just so I understand you:  are you saying that you are looking at further changes to the SDK so that it works with devices that do not support connection/disconnection detection?

I have put dummy functions into my BSP wifi module so that it does not have compile errors.  I have tried to onboard a device with these dummy functions and it seemed to work, but for one of my other users it did not work.  

For the **register_event** function, I only return IOT_ERROR_BAD_REQ.  For **clear_event** function I return nothing.

jy0209-kim, 2022-01-07:
Dear,

> Just so I understand you: are you saying that you are looking at further changes to the SDK so that it works with devices that do not support connection/disconnection detection?
Yes. The SDK will be updated to work with devices that do not support connection/disconnection detection.

In the meanwhile, please try as following for dummy functions in your BSP wifi module.
For the register_event function, please return IOT_ERROR_NONE. For clear_event function, nothing.

BR, Jiyoung.

toddaustin07, 2022-01-08:
OK thanks.  I will try returning those values for now from my dummy functions.

junyoun-kim, 2023-02-15:
@toddaustin07  Do you have more questions about this thread? If not, it might be appreciated if you kindly close this issue.

closed 2023-02-16 by toddaustin07