SmartThingsCommunity/st-device-sdk-c #138: Error: Direct-connected device does not generate the network

huuhuannt1998, 2024-01-19:
I am testing out the Direct-connected device. However, when I tried to connect to the device using Smartthings app, it asked to join a custom network but it was unable to join. How can I fix this?

[image]
[image]

junyoun-kim, 2024-01-30:
@huuhuannt1998 Sorry for late response.

Could you share device's log and SmartThings app version?

Also could you check if you can see same wifi ssid name(Switch Demo_E40BLe0016HedO1UTw) in wifi scan list in phone setting?

junyoun-kim, 2024-11-27:
Hello, @huuhuannt1998   Is there any more questions? We will close this issue if no more updates.

closed 2024-12-02 by junyoun-kim