SmartThingsCommunity/st-device-sdk-c #140: ESP32 switch example fail to start conection to st

Andorezu, 2024-04-13:
Good day

I tried to run the ESP32 switch example but when I pressed for 5 seconds the button the next message appeared.

W (45182) [IoT]: st_conn_start(1226) > Can't start it, iot_main_task is already working(2)

How can I solve this issue?

Below I add all the information the monitor shows.

`I (27) boot: ESP-IDF v5.0.6 2nd stage bootloader
I (27) boot: compile time 15:56:16
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v1.0
I (34) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 stnv             WiFi data        01 02 00009000 00004000
I (72) boot:  1 nvs              WiFi data        01 02 0000d000 00004000
I (79) boot:  2 otadata          OTA data         01 00 00011000 00002000
I (87) boot:  3 phy_init         RF data          01 01 00013000 00001000
I (94) boot:  4 nvs_key          NVS keys         01 04 00014000 00001000
I (101) boot:  5 ota_0            OTA app          00 10 00020000 00150000
I (109) boot:  6 ota_1            OTA app          00 11 00170000 00150000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=33a88h (211592) map
I (206) esp_image: segment 1: paddr=00053ab0 vaddr=3ffb0000 size=03918h ( 14616) load
I (212) esp_image: segment 2: paddr=000573d0 vaddr=40080000 size=08c48h ( 35912) load
I (227) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=bd6e4h (775908) map
I (507) esp_image: segment 4: paddr=0011d70c vaddr=40088c48 size=0dd24h ( 56612) load
I (542) boot: Loaded app from partition at offset 0x20000
I (542) boot: Disabling RNG early entropy source...
I (554) cpu_start: Multicore app
I (554) cpu_start: Pro cpu up.
I (554) cpu_start: Starting app cpu, entry point is 0x400813b8
0x400813b8: call_start_cpu1 at /home/bernardo/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (544) cpu_start: App cpu up.
I (572) cpu_start: Pro cpu start user code
I (572) cpu_start: cpu freq: 160000000 Hz
I (572) cpu_start: Application information:
I (577) cpu_start: Project name:     main
I (582) cpu_start: App version:      v1.8.0-74-gb5fc399-dirty
I (588) cpu_start: Compile time:     Apr 13 2024 15:56:10
I (594) cpu_start: ELF file SHA256:  9249ef378e2e5d4d...
I (600) cpu_start: ESP-IDF:          v5.0.6
I (605) cpu_start: Min chip rev:     v0.0
I (610) cpu_start: Max chip rev:     v3.99 
I (615) cpu_start: Chip rev:         v1.0
I (620) heap_init: Initializing. RAM available for dynamic allocation:
I (627) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (633) heap_init: At 3FFB7C88 len 00028378 (160 KiB): DRAM
I (639) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (645) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (652) heap_init: At 4009696C len 00009694 (37 KiB): IRAM
I (659) spi_flash: detected chip: generic
I (663) spi_flash: flash io: dio
I (668) app_start: Starting scheduler on CPU0
I (672) app_start: Starting scheduler on CPU1
mI (672) main_task: Started on CPU0
I (682) main_task: Calling app_main()
I (692) [IoT]: _dump_device_info(746) > firmware_version: 1.0
I (702) [IoT]: iot_bsp_wifi_init(251) > [esp32] iot_bsp_wifi_init
I (712) wifi:wifi driver task: 3ffc21e4, prio:23, stack:6656, core=0
I (712) system_api: Base MAC address is not set
I (712) system_api: read default base MAC address from EFUSE
I (722) wifi:wifi firmware version: dbb7b72
I (722) wifi:wifi certification version: v7.0
I (722) wifi:config NVS flash: enabled
I (732) wifi:config nano formating: disabled
I (732) wifi:Init data frame dynamic rx buffer num: 32
I (742) wifi:Init static rx mgmt buffer num: 5
I (742) wifi:Init management short buffer num: 32
I (742) wifi:Init dynamic tx buffer num: 32
I (752) wifi:Init static rx buffer size: 1600
I (752) wifi:Init static rx buffer num: 10
I (762) wifi:Init dynamic rx buffer num: 32
I (762) wifi_init: rx ba win: 6
I (762) wifi_init: tcpip mbox: 32
I (772) wifi_init: udp mbox: 6
I (772) wifi_init: tcp mbox: 6
I (782) wifi_init: tcp tx win: 5744
I (782) wifi_init: tcp rx win: 5744
I (782) wifi_init: tcp mss: 1440
I (792) wifi_init: WiFi IRAM OP enabled
I (792) wifi_init: WiFi RX IRAM OP enabled
I (802) [IoT]: iot_bsp_wifi_init(292) > [esp32] iot_bsp_wifi_init done
I (802) [IoT]: st_conn_init(1134) > stdk_version : 1.8.15
I (812) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (822) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (832) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (842) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| IntrI (2862) [IoT]: st_conn_start(1222) > st_conn_start start (no-pin)
I (2862) [IoT]: st_conn_start(1274) > st_conn_start done (0)
I (2862) main_task: Returned from app_main()
I (2862) [IoT]: _do_iot_main_command(586) > curr_main_cmd:0, curr_main_state:0
I (2872) [IoT]: _do_state_updating(435) > current state 0, new state 2
I (2882) [IoT]: iot_bsp_wifi_set_mode(310) > iot_bsp_wifi_set_mode = 1
I (2892) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (2982) wifi:mode : sta (34:ab:95:75:cd:44)
I (2982) wifi:enable tsf
I (2982) [IoT]: esp_wifi_event_post_to_user(106) > Station started
I (2982) [IoT]: iot_bsp_wifi_set_mode(341) > WiFi Station Started
I (5402) [IoT]: esp_wifi_event_post_to_user(185) > event_handler = 1
I (5492) [IoT]: iot_easysetup_create_ssid(89) > >> Bacnight[m1Nc] <<
I (5492) [IoT]: iot_bsp_wifi_set_mode(310) > iot_bsp_wifi_set_mode = 3
I (5492) wifi:mode : sta (34:ab:95:75:cd:44) + softAP (34:ab:95:75:cd:45)
I (5502) wifi:Total power save buffer number: 16
I (5502) wifi:Init max length of beacon: 752/752
I (5512) wifi:Init max length of beacon: 752/752
I (5762) wifi:Total power save buffer number: 16
I (5762) [IoT]: esp_wifi_event_post_to_user(147) > SYSTEM_EVENT_AP_START
I (5762) [IoT]: iot_bsp_wifi_set_mode(470) > AP Mode Started
I (5772) [IoT]: iot_easysetup_init(488) > IOT_STATE_PROV_ES_START
I (5762) [IoT]: esp_wifi_event_post_to_user(152) > SYSTEM_EVENT_AP_STOP
I (5772) [IoT]: es_http_init(159) > http tcp init!!
I (5782) [IoT]: esp_wifi_event_post_to_user(147) > SYSTEM_EVENT_AP_START
I (5792) [IoT]: iot_easysetup_init(524) > IOT_STATE_PROV_ES_INIT_DONE
I (5802) [IoT]: _do_state_updating(533) > Current timeout : 300000 for 0
status: 2, stat: 1
I (5812) [IoT]: _do_state_updating(573) > Call usr status_cb with 0x102
Button event, val: 0, tick: 3946
Button long press, iot_status: 2
I (45132) [IoT]: st_conn_cleanup(1305) > st_conn_cleanup start (0)
W (45132) [IoT]: _delete_dev_card_by_usr(377) > not connected, so can't send device_card deleting msg
I (45132) [IoT]: http_cleanup_all_connection(39) > close listen socket
I (45142) [IoT]: es_http_deinit(184) > http tcp deinit complete!
I (45152) [IoT]: iot_easysetup_deinit(560) > IOT_STATE_PROV_ES_DONE
I (45152) [IoT]: iot_device_cleanup(1023) > start to erase device information
I (45172) [IoT]: st_conn_cleanup(1324) > st_conn_cleanup done (0)
I (45172) [IoT]: st_conn_start(1222) > st_conn_start start (no-pin)
W (45182) [IoT]: st_conn_start(1226) > Can't start it, iot_main_task is already working(2)
fail to start connection. err:-1`

junyoun-kim, 2024-04-15:
@Andorezu Thank  you for having interest in the sdk. We are currently looking into the issue.

Meanwhile could you let me know what you are trying to do? If you were following the instruction on ST app while onboarding, actually you don't have to press the button 5 seconds. By just powering on esp32, the device is already ready for onboarding state.
So when you see the instruction saying pressing the button 5 seconds on ST app, please ignore it now.

Anyway we will prepare the patch for the issue. Thank you for reporting.

Andorezu, 2024-04-15:
@junyoun-kim thank you for your reply.

I made a mistake with my ST app on my devices, it was not updated, after updating the apps on my devices everything works great and you are right, no need to press the button.

junyoun-kim, 2024-11-27:
Hello, @Andorezu  Is there any more questions? We will close this issue if no more updates.

closed 2024-12-02 by junyoun-kim