SmartThingsCommunity/st-device-sdk-c #149: No AP bug is detected during device power-on

thinkaName, 2025-03-06:
After successful pairing.Power off AP.Power on the device again.Power on the AP later.D

evice can't switch the wifi mode

thinkaName, 2025-03-06:
[log.TXT](https://github.com/user-attachments/files/19107108/log.TXT)

junyoun-kim, 2025-03-11:
@thinkaName Which version or branch are you testing on? and which example you are trying to do?

thinkaName, 2025-03-11:
> [@thinkaName](https://github.com/thinkaName) Which version or branch are you testing on? and which example you are trying to do?

v2.0  example switch

junyoun-kim, 2025-03-11:
@thinkaName Could you try switch_example in [reference repo](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref)?

thinkaName, 2025-03-11:
> [@thinkaName](https://github.com/thinkaName) Could you try switch_example in [reference repo](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref)?

I don't think it's a problem with example, it's a bug in the sdk, and example just adds different capabilities

junyoun-kim, 2025-03-11:
The example is outdated, and we'll update the example soon. We no longer support onboarding(provisiong) method the example is using. So please test new example when it's updated.

junyoun-kim, 2025-04-16:
@thinkaName New version(v2.1.1) released. Could you try out again?

junyoun-kim, 2025-04-30:
No more activity

closed 2025-04-30 by junyoun-kim