SmartThingsCommunity/st-device-sdk-c #151: Can I give you an example of hub integration?

butaikis, 2025-03-13:
Hi. Is there any example of the integration of a hub that controls some devices, is there a possibility?

junyoun-kim, 2025-03-13:
@butaikis Hello, Which scenario exactly are you willing to tell?

butaikis, 2025-03-13:
@junyoun-kim I have a device hub that manages different kinds of devices. For example, dimmers, smart sockets, etc. The user adds these devices to the hub via the web interface. Is it possible to support this type of hub and its devices in SmartThings? Maybe there are code examples and settings in the Develop Workspace?

butaikis, 2025-03-13:
@junyoun-kim That's right, I already have a working hub device that supports various devices from the manufacturer. And I can't decide how to integrate them into SmartThings. Via direct connection or cloud.
But the second question is, are such devices generally supported in SmartThings?

junyoun-kim, 2025-03-13:
@butaikis So in summary, you are developing hub devices that support multiple devices. and want to integrate them into SmartThings. I think currently we don't support those type devices. But if you construct your own cloud service with the hub, you can integrate them into SmartThings as [cloud connected devices](https://developer.smartthings.com/devices/cloud-connected).

butaikis, 2025-03-13:
@junyoun-kim Yes, we have our own cloud. Then I will try to integrate through the cloud. I hope to find some examples)

butaikis, 2025-03-15:
@junyoun-kim Maybe you can tell me about another question. I started testing the platform, that's the problem. The Develop Workspace has device profiles, you can create your own profiles and even visually customize them. 
And when I change the json file and save the profile, the devices in the SmartThings app can update immediately, or they can only after some time. How can I force devices to be updated immediately?

[image]

junyoun-kim, 2025-03-17:
Hello @butaikis As I know, you can force them updated by removing SmartThings App cache data. You can eliminate them in phone Setting App menu. Maybe you should enable developer mode again after removing cache.

butaikis, 2025-03-17:
@junyoun-kim I have already deleted the application cache and cleared the data completely, but this does not help. And now, the mini card is displayed as it should be, and the full card is an old sample.

butaikis, 2025-03-17:
@junyoun-kim Deleting components in the profile helps. But only changing the json file apparently does not help.

butaikis, 2025-03-17:
@junyoun-kim json file

```
"dashboard": {
    "states": [
      {
        "component": "main",
        "capability": "switch",
        "version": 1,
        "idx": 0,
        "group": "main",
        "composite": false
      }
    ],
    "actions": [
      {
        "component": "main",
        "capability": "switch",
        "version": 1,
        "idx": 0,
        "group": "main"
      }
    ]
```

**Components**

[image]

How it is displayed:

[image]

junyoun-kim, 2025-03-19:
@butaikis you are saying device-config json file and the displayed ones is different? In above picture, you captured "dashboard" content Actually "detailView" is related with the displaed on ST app device plugin page.

butaikis, 2025-03-19:
@junyoun-kim All right. 

I changed the device profile, first I added 2 switches to the display, created an additional component and changed the JSON. In the SmartThings app, my device immediately changed to what I need. 

Then I wanted to return the device to the standard display. I changed the JSON, uploaded it and nothing happened. It only helps to create a new profile and add devices from the cloud again, but with a different profile. 
It doesn't change in any other way.
I don't think it should be like this.

junyoun-kim, 2025-03-19:
@butaikis I think it's due to server side cache. Could you check it reflect your change after some times like few hours ?

butaikis, 2025-03-19:
@junyoun-kim It's been more than a day)

junyoun-kim, 2025-03-19:
@butaikis Could you share your vid and mnid for the device config json?

butaikis, 2025-03-19:
@junyoun-kim 

**default view**
[deviceConfigST_aca02747-0948-45f4-9014-923c4ddf13aa.json](https://github.com/user-attachments/files/19335432/deviceConfigST_aca02747-0948-45f4-9014-923c4ddf13aa.json)

added an additional component to the display
[modific.json](https://github.com/user-attachments/files/19335484/modific.json)

junyoun-kim, 2025-03-19:
@butaikis What do you want to archive after modification? It looks like you put another capability under "dashboard" and it's invalid

butaikis, 2025-03-19:
@junyoun-kim the file that is **modified.json** - it works, displays as needed.
Then, I upload the standard **deviceConfigST_aca02747-0948-45f4-9014-923c4ddf13aa.json** and nothing happens, the interface doesn't change. As there were buttons with **modific.json**, so they remained.

junyoun-kim, 2025-03-19:
@butaikis Could you share display after modified.json?

junyoun-kim, 2025-03-19:
@butaikis I think that's not a result of modified json. That displayed one is related with "defatilView".

```
  "detailView": [
    {
      "component": "main",
      "capability": "switch",
      "version": 1,
      "values": [],
      "patch": []
    },
    {
      "component": "comp1",
      "capability": "switch",
      "version": 1,
      "values": [],
      "patch": []
    }
  ],
```

Then, what display do you expect after come back with deviceConfigST_aca02747-0948-45f4-9014-923c4ddf13aa.json?

butaikis, 2025-03-19:
@junyoun-kim it's quite strange, because **detailView** I didn't change it, I only changed the **dashboard**. Why is that?

junyoun-kim, 2025-03-19:
@butaikis Currently the displayed one is working correctly as it's configured in json. So it;s looks like working well.

butaikis, 2025-03-19:
@junyoun-kim I deleted the section back in the **detailView**, but it still [displays ](https://github.com/SmartThingsCommunity/st-device-sdk-c/issues/151#issuecomment-2735664022) the same way)

butaikis, 2025-03-19:
@junyoun-kim Correct me if I'm wrong, but.

Initially, I created a profile, added many different components (besides the main one), but they are not displayed in the SmartThings application. Then I wanted to display some components, edit the json, upload and everything is displayed. Ok.

Next, I need to return to the original appearance. I edit the json, but it doesn't help, and I have to delete the component altogether so that everything returns as it was. Why is this so if only the json file affects the display? It's not entirely logical.

junyoun-kim, 2025-03-25:
@butaikis There are some server cache issue about reflecting updated content to SmartThings app. So there might be a delay for updating UI.

So, If you want to update basic component and Capabilites configuration, then editing under Component & Capabilties tab is enough. But it will take some time to reflect to SmartThings app.

We are fixing server cache issue. but it will take some times.

junyoun-kim, 2025-04-16:
@butaikis Is there any more questions? Please help me close the issue.

butaikis, 2025-04-16:
@junyoun-kim Has the problem been fixed, or is it still in progress?

butaikis, 2025-04-16:
@junyoun-kim can I find out about the language settings?

junyoun-kim, 2025-04-16:
@butaikis  If you are using Android, you can change language setting in Setting -> General management -> Language. You can change any other language other than current language and delete SmartThings app cache and data. Then you can get updated content.

butaikis, 2025-04-16:
@junyoun-kim That's what I thought) I decided to clarify if I got it wrong. We will wait for the fix on the server, thank you.

closed 2025-04-16 by butaikis