SmartThingsCommunity/st-device-sdk-c #152: BLE link disconnects with HCI reason 0x13 immediately after “send complete”(ESP32-C6)

darzento, 2025-04-16:
| Item | Value |
|------|-------|
| MCU / Module | ESP32‑C6 |
| ESP‑IDF | v5.4.0 |
| SmartThings Device SDK | v2.1.1 |
| Phone | Galaxy S10e |

I (54550) [IoT]: es_msg_disassemble(292) > send complete
I (54603) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (54603) BLE_ONBOARD: ESP_GATTS_CONF_EVT, status 0x00 attr_handle 42
W (56742) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13 <-- When this message appears, the connection to the smartphone is lost and the registration cannot proceed. What should I do?

Below is the log captured after the smartphone connection. Please have a look.

```
I (52644) BT_BTM: BTM_InqDbRead: bd addr [4773d2701d4d]

NOTHING!
main_task_tick
I (52729) BT_GATT: GATT_GetConnIdIfConnected status=1

I (52730) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (52731) BLE_ONBOARD: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 47:73:d2:70:1d:4d:
I (52739) [IoT]: _iot_easysetup_ble_event_cb(93) > BLE Gatt Connection
I (52748) BT_APPL: BTA_DmBleBroadcast: start = 0 

I (52753) BLE_ONBOARD: Stop adv successfully

I (52827) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (52970) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53068) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53165) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53249) BLE_ONBOARD: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 30,latency = 0, timeout = 400
I (53324) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53398) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53473) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53548) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53590) BLE_ONBOARD: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 6,latency = 0, timeout = 500
I (53597) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53619) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53634) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
NOTHING!
main_task_tick
I (53657) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53672) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53687) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53701) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53724) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53762) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (53762) BLE_ONBOARD: GATT_WRITE_EVT, conn_id 0, trans_id 1, handle 43 need resp 1 is_prep 0
I (53766) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 1  Status: 0x0000

I (53773) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53791) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53793) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (53795) BLE_ONBOARD: ESP_GATTS_MTU_EVT, MTU 185
I (53855) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (53856) BLE_ONBOARD: GATT_WRITE_EVT, conn_id 0, trans_id 2, handle 42 need resp 1 is_prep 0
I (53859) [IoT]: es_msg_assemble(126) > op_code : 0, cmd_num : 1, transaction_id : 0, chunk_data_continued : 0, total_size : 55
I (53871) [IoT]: es_msg_assemble(187) > msg command num[1]
I (53877) [IoT]: es_msg_assemble(194) > msg transfer is completed
I (53884) [IoT]: es_msg_assemble(211) > es_msg_dispatch is called
I (53891) BLE_ONBOARD: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 30,latency = 0, timeout = 400
I (53903) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 2  Status: 0x0000

I (53910) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4773d2701d4d
I (53920) [IoT]: _es_process_accepted_connection(72) > ble event reported
I (53926) [IoT]: iot_easysetup_ble_msg_handler(391) > cmd : 0
NOTHING!
main_task_tick
[... 100 lines cut ...]
se_storage_write
esp_pm_lock_release success
esp_pm_lock_delete success

 . uone_i2c_deinit Success : SW = 0x00
esp_pm_lock_create success
esp_pm_lock_acquire success
I (57081) gpio: GPIO[7]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (57091) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

 << UONE Driver Version : 010000 >>
esp_pm_lock_release success
esp_pm_lock_delete success

 . uone_i2c_deinit Success : SW = 0x00
I (57230) BT_APPL: BTA_DmSetBleAdvParamsAll: 32, 64

I (57233) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0

I (57246) BLE_ONBOARD: Start adv successfully

I (57247) [IoT]: _do_iot_main_command(574) > curr_main_cmd:0, curr_main_state:0
I (57255) [IoT]: _do_state_updating(361) > current state 0, new state 2
I (57262) [IoT]: iot_bsp_wifi_set_mode(324) > iot_bsp_wifi_set_mode = 1
```

junyoun-kim, 2025-04-16:
@darzento Did you try with latest SmartThings Mobile App? Could you share SmartThings App version? or Do you use iOS or Android?

darzento, 2025-04-16:
@junyoun-kim I used the SmartThings App version Android : v1.8.29.24 and ios : v1.7.29.16

junyoun-kim, 2025-04-16:
@darzento We have iOS SmartThings App issue currently and will be fixed in next release. But are you facing the issue with Android SmartThings App? Do you have any change to run with ESP32 or ESP32S3?

darzento, 2025-04-16:
@junyoun-kim  That log is from when I used Android app. And I'm using esp32c6

junyoun-kim, 2025-04-16:
@darzento Actually we did't test with esp32c6 yet. Could you try other series like esp32?

darzento, 2025-04-16:
@junyoun-kim I don't have any esp32, but I confirmed that v1.8.15 (beta version) works with esp32c6.
I was going to apply version 2.1.1 to get it authenticated, but the previous version worked well too, so I'll roll back and use it.

junyoun-kim, 2025-04-16:
@darzento v1.8.15 version is using softAP between Mobile and device. From v2.0, we've changed from softAP to BLE, becuase it's more recommended D2D communication. Could you create new Project in Developer Workspace and test it again?

junyoun-kim, 2025-04-30:
Hello @darzento Is there any more issue? Can I close this issue?

junyoun-kim, 2025-05-14:
No more activity.

closed 2025-05-14 by junyoun-kim